egnor/wt32-eth01 #3: another way to achieve auto-boot on eth01

cyberboy666, 2023-12-13:
thanks for putting this together! been working with the ETH01 for a few projects - wish i found this lil summary/guide earlier!

just wanted to share the method i found for flashing the board:

at first i was using a generic usb-to-serial chip with RX->TX, TX->RX and manually setting IO0 low then flashing then reseting...

then after reading about [Automatic Bootloader](https://docs.espressif.com/projects/esptool/en/latest/esp32s2/advanced-topics/boot-mode-selection.html) it looks like only need to find a usb-to-serial that has RTS and DTR.

[image]

i ended up getting a [WeMos CH340G Breakout Board](https://www.aliexpress.com/item/4001290559806.html) from aliexpress for under a dollar. the default VCC on this is 5v so i just connect to the 5v input on ETH01 otherwise only needing to connect RTS -> EN and DTR -> GPIO0 to get auto-bootloader for very cheap:

[image]

however since i am using platformio and arduino core with ETH01 i found that while connecting like this, opening the serial monitor disabled the board... this is described on the espressif docs above as:

> Some OS and/or drivers may activate RTS and or DTR automatically when opening the serial port (true only for some serial terminal programs, not esptool.py), pulling them low together and holding the ESP in reset. If RTS is wired directly to EN then RTS/CTS “hardware flow control” needs to be disabled in the serial program to avoid this. An additional circuitry is implemented in order to avoid this problem - if both RTS and DTR are asserted together, this doesn’t reset the chip. The schematic shows this specific circuit with two transistors and its truth table.

based on this and the schematic they show i worked around this issue using two resistors and (2n3904) transistors on breadboard:

[image]
 
adding this as it took me some time to figure this out and hoping it may help someone else getting started with these great little controllers (lemme know if it would be useful to somehow add this info as pull request or feel free to adapt it form readme as u see fit :) )

[image]

egnor, 2023-12-13:
Cool, I'll figure out some way to distill this down for the docs! If nothing else, a pointer to that "Automatic Bootloader" reference is a good thing to have.

Do you have a capacitor in your protoboard implementation? If not, do you find yourself experiencing glitches the way the doc suggests?

I've actually recently designed a "cradle" for the WT32-ETH01 that breaks out the pins into screw terminals, and among other things includes a USB port, USB-serial chip, and the recommended two-transistors-and-a-capacitor circuit. (I used FETs instead of BJTs so I wouldn't need to include resistors.)

cyberboy666, 2023-12-16:
i dont have the cap as they suggest and havnt seen this issue. will add the cap tho just in case!

closed 2023-12-16 by cyberboy666

egnor, 2023-12-16:
Ah, that's probably becasue the WT32-ETH01 already includes a 0.1µF cap on EN! The ESP32 docs recommend a 1-10µF cap for this purpose, but 0.1µF might be enough and if you're lucky you wouldn't get a glitch of this sort anyway.

Anyhoooooo I've [the README to talk about this topic more](https://github.com/egnor/wt32-eth01/blob/main/README.md#programming). I shamelessly copied your protoboard pic (with attribution), let me know if you object to that!

(Closing this bug for my workflow, feel free to reopen with any suggestions or corrections)

cyberboy666, 2023-12-16:
ah yes makes sense ! 

saw that you updated readme - looks great - thank you !