egnor/wt32-eth01 #4: Thanks for the info and some notes

comporder1, 2023-12-17:
Hi, 
First, thank you for taking time to gather and post this info. I was 80% into designing a PCB using the WT32-eth01 for a WLED project when I found this project. It helped has helped answer several questions and given me peace of mind with some of my decisions.  

Biggest problem I had with my project was that IO5 and IO35 are sometimes swapped?? I designed my PCB with the pinout found on the [Amazon listing](https://a.co/d/94t4u4t) that purchased from. It has IO5 beside IO4. You can imagine my surprise when I test my PCB the first time! After googling, I have found several sites show IO5 beside IO4 and many the other way. I think it deserves a note here on your project to be aware of the discrepancies. 

Lastly, I did not have a generic "programmer" on hand when I received my WT32, so I decided to try out an old standby trick that has worked for me in the past. Take a simple arduino uno, remove the microcontroller IC, connect 3v3, gnd, TX->TX, and RX->RX, jump IO1->gnd, and away you go! (notice you don't have to cross TX/RX when you use this method. It is labeled according to microcontroller on the board.)

egnor, 2023-12-18:
Fascinating. Interestingly, the schematic I have actually shows the 2/4/5 layout, not the 2/4/35 layout I see on all the hardware I've met. Have you seen physical hardware with the 2/4/5 layout?

I will make some updates to docs to mention these things!

comporder1, 2023-12-18:
This is my first experience with the WT32. All that I received are 2/4/35. I’m glad I could contribute.

egnor, 2023-12-18:
Updated the README, take a look and re-open this if you have any suggested changes!

closed 2023-12-18 by egnor