egnor/wt32-eth01 #16: Incorrect clock selection

EchterAgo, 2025-01-28:
The change in #14 is not correct. The previous code was using `ETH_CLOCK_GPIO0_IN` which is the default. Changing it to `ETH_CLOCK_GPIO17_OUT` will leave the ESP MAC running off a different clock than the LAN8720, which still runs off the external oscillator.

Using `ETH_CLOCK_GPIO17_OUT` does initially seem to work but it causes periodic packet loss because the two oscillators are not in sync.

GPIO17 is routed to the TX LED / Pin, so you can see the LED glow faintly. It seems to me that `ETH_CLOCK_GPIO0_IN` matches the ESP-IDF example code in `README.md`.

EchterAgo, 2025-01-28:
@ProgrammingJohn

egnor, 2025-01-28:
What would you say the correct settings are?

EchterAgo, 2025-01-29:
`ETH.begin(ETH_PHY_LAN8720, 1, 23, 18, 16, ETH_CLOCK_GPIO0_IN)`

or in the version of the [ESP32 Arduino framework still in use by PlatformIO (< v3.0)](https://github.com/platformio/platform-espressif32/issues/1225):

`ETH.begin(1, 16, 23, 18, ETH_PHY_LAN8720, ETH_CLOCK_GPIO0_IN);`

Alternatively you can use:

```
#define ETH_PHY_POWER 16
ETH.begin()
```

or:

```
pinMode(16, OUTPUT);
digitalWrite(16, HIGH);
ETH.begin()
```

ProgrammingJohn, 2025-01-29:
Apologies with that oversight. Thank you for finding a correction. I simply ran with what worked after looking at some other documentation online but honestly didn't know the effect of the `ETH_CLOCK_GPIO0_IN` argument. I will switch back to `IN`

Regardless my main fix was the updated positional arguments for `ETH.begin`

Thanks again!

EchterAgo, 2025-01-30:
Looks good, closing.

closed 2025-01-30 by EchterAgo