egnor/wt32-eth01 #18: Pin for PWM piezo buzzer

4w4w4w, 2025-01-31:
Hi, do you know which pin I should use for PWM to drive a passive piezo buzzer?

egnor, 2025-02-01:
Almost any general I/O pin should work, I believe!

4w4w4w, 2025-02-01:
it certainly did, here is the esphome device script to program with, there is an action to play 2 piezo buzzers simultaneously or one at a time:

```
esphome:
  name: 
  friendly_name: 
  platform: ESP32
  board: esp-wrover-kit

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

output:
  - platform: ledc
    pin: GPIO12
    id: rtttl_out  

  - platform: ledc
    pin: GPIO15
    id: rtttl2_out  

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: ""
  actions:
    - action: play_rtttl_both
      variables:
        song_str: string
      then:
        - lambda: |-
            id(rtttl_component1)->play(song_str);
            id(rtttl_component2)->play(song_str);

    - action: play_rtttl_1
      variables:
        song_str: string
      then:
        - lambda: |-
            id(rtttl_component1)->play(song_str);

    - action: play_rtttl_2
      variables:
        song_str: string
      then:
        - lambda: |-
            id(rtttl_component2)->play(song_str);

ota:
  - platform: esphome
    password: ""

rtttl:
  - id: rtttl_component1
    output: rtttl_out
    gain: 50%
    on_finished_playback:
      - logger.log: 'Song ended on buzzer 1!'

  - id: rtttl_component2
    output: rtttl2_out
    gain: 50%
    on_finished_playback:
      - logger.log: 'Song ended on buzzer 2!'

```

closed 2025-02-01 by 4w4w4w