egnor/wt32-eth01 #20: Is there a "client" component for this library?

marclefevre, 2025-04-28:
I've been successful in connecting a WT32-eth01 to a network and I'm on to the next step of doing something useful with that connection.  My first step is to make a connection to an MQTT broker on the same network but I'm having trouble figuring out how.  With a WIFI connection, using the Arduino core, I would do something like this:

```
WiFiClient     wifiClient;
PubSubClient   mqttClient(wifiClient);
```
to provide the PubSubClient class a communication Stream for its I/O.  I'm looking for a similar mechanism for Ethernet connections.

I have tried this

```
#include <Ethernet.h>

EthernetClient   ethClient;
PubSubClient     mqttClient(ethClient);
```

which will compile and link but the PubSubClient object cannot make a connection to the broker.  I don't find any examples (perusing the web) that help me so I'm hoping you have some insight on how to get his working.

egnor, 2025-04-29:
I've never done this, but in theory [EthernetESP32 configured for the LAN8720 PHY](https://github.com/Networking-for-Arduino/EthernetESP32?tab=readme-ov-file#phy-modules) and set up for the WT32-ETH01's pins should work and provide a compatible API.

(Vanilla EthernetClient is written for the W5500, so won't work here.)

If you get this working please follow up so we can add the info to the README here!

marclefevre, 2025-04-30:
I saw your reply just as I was trying what you suggested.  It works.  My example was using the client interface to connect to an MQTT broker.  Here is the basics needed to link the relevant parts together:

in main.cpp
```
#include <PubSubClient.h>
#include <EthernetESP32.h>

EMACDriver       driver(ETH_PHY_LAN8720, ETH_PHY_MDC, ETH_PHY_MDIO, ETH_PHY_POWER);
EthernetClient   ethClient;
PubSubClient     mqttClient(ethClient);
```
I'm using platformio and vscode and here are the platformio.ini entries needed to make it work:

```
lib_deps =
   knolleary/PubSubClient@^2.8
   https://github.com/Networking-for-Arduino/EthernetESP32.git
```

egnor, 2025-05-06:
README updated with these notes! Thanks!!

closed 2025-05-06 by egnor